stockmann-lab/Open_Source_MRI_Tools_MGH #53: Document GPA specs and modifications on wiki

lcbMGH, 2020-09-08:


lcbMGH, 2020-09-19:
Where should this be documented? This is the last step for finishing this up.

jaystock, 2020-09-19:
Good question.  Thanks for checking into this.   We might want to put it on the tabletop.martinos.org wiki under the GPA page.   We already have two GPA designs there and this could be added as a third section demonstrating how the GPAs can be used at a higher voltage and with added buffers.  We could also list the ammeter used, etc.  What do you think?  I can create an account for you on the tabletop wiki.   What username would be good?

Jason

        External Email - Use Caution

The information in this e-mail is intended only for the person to whom it is addressed. If you believe this e-mail was sent to you in error and the e-mail contains patient information, please contact the Mass General Brigham Compliance HelpLine at http://www.massgeneralbrigham.org/complianceline . If the e-mail was sent to you in error but does not contain patient information, please contact the sender and properly dispose of the e-mail.

closed 2020-09-30 by lcbMGH